frank-zago/ch341-i2c-spi-gpio #2: no i2c and spi devices.

Vicsar73, 2022-08-29:
I'm using Debian 11. CH341T board. Compiled without errors.

dmesg
[ 1873.605531] usb 1-2: New USB device found, idVendor=1a86, idProduct=5512, bcdDevice= 3.04
[ 1873.605534] usb 1-2: New USB device strings: Mfr=0, Product=0, SerialNumber=0

sudo insmod ch341-buses.ko
ls -l /dev/* | grep ch341
crw-rw---- 1 root ch341   254,   0 сер 29 14:47 /dev/gpiochip0

why don't i2c and spi devices show up?

frank-zago, 2022-08-29:
Not sure for i2c. "i2cdetect -l" should show all i2c devices, including that one.

As for SPI, the ch341T doesn't physically expose it. The spidev driver would need to be loaded too, and a device added (see README).

joeyoung, 2022-12-05:
I have also observed that the i2c and spi do not appear in /dev, either before or after insmod the driver. But on plugging in the module a gpiochip2 is added to the list. 
I tried adding the blacklist in case the ch341serial driver was conflicting. It made no difference to the i2c mode, but the blacklist does stop the module being recognized as a ttyUSBx when the module is jumper configured for serial mode.
I am using LMDE5 which appears to have some i2c capability already as shown below, so perhaps the driver conflicts with the built-in driver(s)?
`jyoung@work:~$ tree /sys/bus/i2c
/sys/bus/i2c
├── devices
│   ├── i2c-0 -> ../../../devices/pci0000:00/0000:00:0a.1/i2c-0
│   ├── i2c-1 -> ../../../devices/pci0000:00/0000:00:0a.1/i2c-1
│   ├── i2c-2 -> ../../../devices/pci0000:00/0000:00:05.0/i2c-2
│   ├── i2c-3 -> ../../../devices/pci0000:00/0000:00:05.0/i2c-3
│   └── i2c-4 -> ../../../devices/pci0000:00/0000:00:05.0/i2c-4
├── drivers
│   ├── CHT Whiskey Cove PMIC
│   │   ├── bind
│   │   ├── uevent
│   │   └── unbind
│   ├── dummy
│   │   ├── bind
│   │   ├── uevent
│   │   └── unbind
│   └── intel_soc_pmic_i2c
│       ├── bind
│       ├── uevent
│       └── unbind
├── drivers_autoprobe
├── drivers_probe
└── uevent

10 directories, 12 files
`

frank-zago, 2022-12-06:
Maybe you're having some udev issue. Here's what I get when plugging the device:

    #sudo udevadm monitor
    monitor will print the received events for:
    UDEV - the event which udev sends out after rule processing
    KERNEL - the kernel uevent

    KERNEL[7410.040242] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3 (usb)
    KERNEL[7410.046945] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/3-3:1.0 (usb)
    KERNEL[7410.047186] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/3-3:1.0/i2c-11/i2c-dev/i2c-11 (i2c-dev)
    KERNEL[7410.047221] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/3-3:1.0/i2c-11 (i2c)
    KERNEL[7410.047403] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/gpiochip2 (gpio)
    KERNEL[7410.047437] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/gpio/gpiochip744 (gpio)
    KERNEL[7410.047513] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/3-3:1.0/spi_master/spi0 (spi_master)
    KERNEL[7410.051479] bind     /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/3-3:1.0 (usb)
    KERNEL[7410.051541] bind     /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3 (usb)
    UDEV  [7410.084874] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3 (usb)
    UDEV  [7410.087948] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/gpio/gpiochip744 (gpio)
    UDEV  [7410.089902] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/3-3:1.0 (usb)
    UDEV  [7410.090072] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/gpiochip2 (gpio)
    UDEV  [7410.091669] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/3-3:1.0/i2c-11/i2c-dev/i2c-11 (i2c-dev)
    UDEV  [7410.092979] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/3-3:1.0/i2c-11 (i2c)
    UDEV  [7410.093663] add      /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/3-3:1.0/spi_master/spi0 (spi_master)
    UDEV  [7410.097455] bind     /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3/3-3:1.0 (usb)
    UDEV  [7410.111020] bind     /devices/pci0000:00/0000:00:07.1/0000:2a:00.3/usb3/3-3 (usb)
    KERNEL[7410.143951] add      /devices/parport0/ppdev0.0 (parport)
    KERNEL[7410.143991] bind     /devices/parport0/ppdev0.0 (parport)
    KERNEL[7410.144011] unbind   /devices/parport0/ppdev0.0 (parport)
    KERNEL[7410.144030] remove   /devices/parport0/ppdev0.0 (parport)
    UDEV  [7410.145041] add      /devices/parport0/ppdev0.0 (parport)
    UDEV  [7410.145539] bind     /devices/parport0/ppdev0.0 (parport)
    UDEV  [7410.145984] unbind   /devices/parport0/ppdev0.0 (parport)
    UDEV  [7410.146383] remove   /devices/parport0/ppdev0.0 (parport)

kumy, 2023-03-17:
Hi,

I have quite a similar issue, I'm not able to get the `/dev/spidevX.Y` devices created (except that the `i2c-14` device is created in my case)

After loading modules:
```
# sudo modprobe spidev
# sudo insmod ch341-buses.ko
```

I plug the device, `dmesg` show:
```
# dmesg

[ 2460.134662] usb 3-11.2.5.3: new full-speed USB device number 15 using xhci_hcd
[ 2460.236022] usb 3-11.2.5.3: New USB device found, idVendor=1a86, idProduct=5512, bcdDevice= 3.04
[ 2460.236032] usb 3-11.2.5.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
```

`udevadm monitor` show
```
# sudo udevadm monitor

monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[2460.178559] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3 (usb)
KERNEL[2460.187900] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0 (usb)
KERNEL[2460.188179] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0/i2c-14/i2c-dev/i2c-14 (i2c-dev)
KERNEL[2460.188223] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0/i2c-14 (i2c)
KERNEL[2460.188446] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/gpiochip1 (gpio)
KERNEL[2460.188520] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/gpiochip1 (gpio)
KERNEL[2460.188597] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/gpio/gpiochip545 (gpio)
KERNEL[2460.188799] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0/spi_master/spi0 (spi_master)
KERNEL[2460.189046] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0 (usb)
KERNEL[2460.189099] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3 (usb)
UDEV  [2460.200577] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3 (usb)
UDEV  [2460.201546] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0 (usb)
UDEV  [2460.201731] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/gpiochip1 (gpio)
UDEV  [2460.201859] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/gpio/gpiochip545 (gpio)
UDEV  [2460.202296] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0/i2c-14/i2c-dev/i2c-14 (i2c-dev)
UDEV  [2460.202315] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/gpiochip1 (gpio)
UDEV  [2460.202779] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0/spi_master/spi0 (spi_master)
UDEV  [2460.202796] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0/i2c-14 (i2c)
UDEV  [2460.203396] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0 (usb)
UDEV  [2460.336114] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3 (usb)
```

And some devices are created:
```
# ll /dev/spi*; ll /dev/|grep ch341
ls: cannot access '/dev/spi*': No such file or directory
crw-rw----   1 root ch341   254,     1 Mar 17 07:47 gpiochip1
crw-rw----   1 root ch341    89,    14 Mar 17 07:47 i2c-14
```

Then I try to add a new device:
```
# sudo bash -c 'echo "spidev 1" > /sys/class/spi_master/spi0/new_device'
```

There is nothing new in `dmeg` output, and `udevadm monitor` show 2 new lines:
```
KERNEL[2594.866588] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0/spi_master/spi0/spi0.1 (spi)
UDEV  [2594.870776] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0/spi_master/spi0/spi0.1 (spi)
```

A new directory tree is created in `/sys/class`
```
[... 124 lines cut ...]
		79:1C:AA:CD:99:E4:B4:4A:F6:7E:B3:85:D4:4A:9C:FC:F2:27:A2:73:
		8E:A8:83:3E:27:30:35:4D:BC:75:23:C3:A6:B9:01:5B:51:55:87:5B:
		63:81:3B:2C:11:B2:7E:1F:85:BB:4C:AC:87:D3:9B:15:0E:6A:EE:AB:
		D8:FE:34:FD:19:DD:63:53:9C:F3:54:9F:26:F4:82:5E:2D:5F:57:90:
		61:00:E4:0B:81:40:80:8C:6E:30:F9:60:93:90:2E:CA:87:AE:E6:F5:
		74:17:FE:1C:BD:E2:E8:37:3D:70:71:E8:58:F5:E6:38:6E:53:7C:F7:
		A2:04:EA:73:F1:A8:15:48:3F:25:D6:5A:50:72:A7:3F:C5:E9:94:70:
		AE:31:3A:41:7B:01:B0:01:4D:E8:95:CC:B0:75:7C:79:1E:5A:2C:35:
		C5:40:46:4F:F8:FD:7D:AA:8E:FD:61:54:E0:7D:C2:11:27:3C:41:D1:
		A8:D8:E2:59:01:10:BD:9D:B7:04:6D:70:9D:62:D6:4E:84:88:50:44:
		9B:86:08:19:65:E7:15:02:AE:56:A2:4F:28:87:1E:55:62:15:11:4E:
		FD:7A:A3:F8:49:3F:9F:78:10:09:41:D5:2A:93:E4:73:F5:20:37:49:
		25:EF:55:C6:30:CA:99:77:13:E0:F4:56:3C:75:39:72:C2:AB:73:73:
		FE:95:B2:C5:21:D3:DD:9F:79:BD:7C:05:DA:50:7B:C4:1A:F6:AB:64:
		91:EF:2E:08:A9:5A:72:36:BF:3A:C5:1E:1D:69:E0:33:8B:A3:6B:7A:
		A8:AF:F5:D4:48:9B:7E:6F:2D:D8:29:95:70:6B:92:0B:78:1E:A7:E6:
		12:31:09:5F:BA:F9:51:2E:29:10:C7:46:24:00:A0:22:A9:01:B2:23:
		77:03:AC:93:22:A4:C2:7F:A1:68:61:2F:ED:75:21:5B:70:00:C6:D2:
		72:A7:B8:2C:04:81:F1:11:12:DA:A1:09:11:44:6B:68:99:AB:B4:C1:
		84:DE:EF:BA:C3:04:FA:26:01:85:32:AA
parm:           bufsiz:data bytes in biggest supported SPI message (uint)
```

Am I missing something?
Thanks for the module and for your help :)

kumy, 2023-03-17:
From that post https://github.com/gschorcht/spi-ch341-usb/issues/22#issuecomment-1055882095 I can have the device created:

```
$ sudo bash -c 'echo spidev > /sys/bus/spi/devices/spi0.1/driver_override'
$ sudo bash -c 'echo spi0.1 > /sys/bus/spi/drivers/spidev/bind'

$ ll /dev/spi*; ll /dev/|grep ch341
crw-rw---- 1 root ch341 153, 0 Mar 17 08:28 /dev/spidev0.1
crw-rw----   1 root ch341   254,     1 Mar 17 08:27 gpiochip1
crw-rw----   1 root ch341    89,    14 Mar 17 08:27 i2c-14
crw-rw----   1 root ch341   153,     0 Mar 17 08:28 spidev0.1

$ gpioinfo gpiochip1
gpiochip1 - 16 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed        "CS1"  output  active-high [used]
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed        "SCK"  output  active-high [used]
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       "MOSI"  output  active-high [used]
	line   6:      unnamed       unused   input  active-high 
	line   7:      unnamed       "MISO"   input  active-high [used]
	line   8:      unnamed       unused   input  active-high 
	line   9:      unnamed       unused   input  active-high 
	line  10:      unnamed       unused   input  active-high 
	line  11:      unnamed       unused   input  active-high 
	line  12:      unnamed       unused   input  active-high 
	line  13:      unnamed       unused   input  active-high 
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high

# udevadm monitor
UDEV  [4900.508493] add      /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0/spi_master/spi0/spi0.1/spidev/spidev0.1 (spidev)
UDEV  [4900.509056] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-11/3-11.2/3-11.2.5/3-11.2.5.3/3-11.2.5.3:1.0/spi_master/spi0/spi0.1 (spi)

```

Is that manual action expected?

frank-zago, 2023-03-19:
Thanks for finding out the problem. Yes, this is a change that looks needed now. From the kernel documentation:

    Sysfs also supports userspace driven binding/unbinding of drivers to
    devices that do not bind automatically using one of the tables above.
    To make the spidev driver bind to such a device, use the following:
    
        echo spidev > /sys/bus/spi/devices/spiB.C/driver_override
        echo spiB.C > /sys/bus/spi/drivers/spidev/bind

Quite annoying, but can we do???

frank-zago, 2023-04-18:
The issue has been found and it's now documented in the README, I can close this ticket.

closed 2023-04-18 by frank-zago