frank-zago/ch341-i2c-spi-gpio #26: udev for SPI

jp-bennett, 2024-01-08:
I'm trying to put together udev rules for firing up an SPI device automatically on plug. I think there's an issue, that the ch341-spi driver doesn't actually expose what SPI number the USB has grabbed. So, on most systems, that will be "/sys/class/spi_master/spi0" but if the system has an existing SPI bus, I suspect the CH341 may become SPI1 instead. Would it be possible to set a udev ENV value that identifies which SPI bus number the driver is using? Or is there an easier solution?

frank-zago, 2024-01-09:
This is not managed by the driver, but the spi subsystem. Run "udevadm monitor", plug the device, and see if you get what you need.

mjsir911, 2024-03-09:
https://github.com/rogerjames99/spi-ch341-usb & https://github.com/rogerjames99/spi-ch341-usb/blob/master/udev/90-ch341-spi.rules seems to work *somewhat*.

After copying that file to `/etc/udev/rules.d/`, all I need to do to set up the device is:
```
echo spidev 0 | sudo tee /sys/class/spi_master/spi1/new_device
```