frank-zago/ch341-i2c-spi-gpio #36: Driver doesn't support 0-length transfers to control CS line

jp-bennett, 2024-04-15:
We use a 0-length ioctl transfer to control the CS:
```
    xfer.tx_buf = NULL;
    xfer.len = 0;

    xfer.rx_buf = NULL
    xfer.cs_change = true; //(or false)
```
This is returning a 110 timeout error. Instead, it should either enable or disable the CS line.

frank-zago, 2024-08-21:
A different driver for SPI only has been merged in the kernel 6.11 (spi-ch341.ko). Maybe this one will work.

jp-bennett, 2024-08-21:
> A different driver for SPI only has been merged in the kernel 6.11 (spi-ch341.ko). Maybe this one will work.

Exciting that a driver has made the kernel. The only downside is it doesn't expose the other pins as GPIO. Thanks for the info!

yezhonghui2024, 2025-03-26:
https://github.com/frank-zago/ch341-i2c-spi-gpio/pull/43 can fix it

frank-zago, 2025-04-12:
merged

closed 2025-04-12 by frank-zago